ajs123/KBikeBLE #1: Parts List

Longhorn17, 2021-06-26:
I'm a fellow M3 bike user and have been looking for this exact solution for my bike to be able to connect via bluetooth with fitness apps.

I'm new to coding and these kind of DIY projects in general and I thought this was a good real world project to dive into but I'm struggling a little to know which components are necessary and compatible for this project.

I was wondering if you could post a list of the exact Arduino board, display, and other parts for this project that you are using?

Sorry if this isn't the correct forum to ask this, I'm new to Github as well.

I think what you have done is pretty awesome and appreciate you sharing it.

nealjane, 2021-06-26:
Hi @Longhorn17 ,

depending where you are In The world- but an Easyer solution if you’re in Europe is the ‘Dpcycling’ m3 adapter- it’s ~£100 imported from Spain to Uk. (Have had one for 1.5years and makes your m3 work like M3i with Bluetooth)

www.siliconegrips.es

Longhorn17, 2021-06-26:
I've come across that before but unfortunately I live in the US. 

I'll look into just upgrading the computer. Thanks for the suggestion!

nealjane, 2021-06-26:
https://m.facebook.com/groups/k2piconverter/permalink/983401195809979/?ref=group_browse

ajs123, 2021-06-28:
I just added a parts list.  I hope that helps!

ajs123, 2021-06-29:
I added a parts list to the archive. There are only four parts:

   - Adafruit nRF52840 Feather Express
   - Generic I2C interface 128x64 OLED display (link to the one I used in
   the parts list)
   - An RJ9 cable to plug into the connector on the bike (cut off the other
   end to connect to the Feather) as shown in the comments in the code)
   - Optionally, a battery so that you don't need to keep USB power in place
   - Some kind of case. Right now, I'm using a small cardboard box with a
   frame cut in the top, a piece of an old CD case as a window, and foam to
   keep the components from rattling around!

 I hope that helps!

Longhorn17, 2021-08-02:
Following up to say thanks! I have been able to get all of the parts to work on their own but am struggling to cut and strip the RJ9 cable to attach to the board. I don't know if the cables I bought are just too cheap but each time I try to strip the insulation off the smaller wires, I end up pulling all of the wiring out too. 

Do you have any recommended tools or advice on the best way to connect the RJ9 cable to the board?

ajs123, 2021-08-02:
It can be hard because those cables are made to be terminated with “insulation displacement connectors” in which the wires are forced into narrow slots that cut through the insulation. A conventional wire stripper worked for me, and with care I was able to twist the conductors with a more robust lead wire, quickly solder, and cover with heat shrink tubing. 

You could go with [something like this…](https://www.amazon.com/Aexit-Accessories-Telephone-Connectors-Connector/dp/B07LFHBCBC)

ajs123, 2021-08-04:
@Longhorn17 - In case it helps... I've started on a HOWTO. The beginning of it is in branch 0.18-working.

Longhorn17, 2021-08-04:
@ajs123 I started to read through the HOWTO and looks very helpful. I'm still working on connecting the RJ9 cable to the feather but hopefully am able to knock that out sometime this week. Thanks!

ajs123, 2021-08-05:
That's great. I'll be adding a bit more as I'm able.

ajs123, 2021-08-07:
@Longhorn17 there are now some additional cable ideas in the HOWTO.

jaredaggie, 2021-11-15:
Does anyone know if this will work with a bike made in 2009? Mine is serial number 090805-10828! Thanks for the awesome work!

jaredaggie, 2021-11-15:
> @ajs123 I started to read through the HOWTO and looks very helpful. I'm still working on connecting the RJ9 cable to the feather but hopefully am able to knock that out sometime this week. Thanks!

We're you able to get this to work on your bike? Thanks!

ajs123, 2021-11-15:
@jaredaggie Yes - It should work exactly the same. If you’re interested in
different ways to connect to the bike, see Build.md for an example using
connectors that mate with the cable in the bike.

Also please note that with v1.1 of the software the default pin assignments
on the feather have changed to make wiring a little bit more
straightforward. I see that I’ve not updated the HOWTO to match. See the
chart at the end of step 7 of Build.md. Of course you can still change
bike_interface.h to match how you chose to wire it, as long as
RESISTANCE_PIN is an analog input. I've added an update of the HOWTO to the
TODO list.

Please don’t hesitate to ask if you get stuck.

or Android