ajs123/KBikeBLE #6: Bluetooth Function only

chanyork, 2022-03-02:
Hello,

1. Is it possible to create a version for us that only add the bluetooth function to the Keiser M3, while keeping the original display?  
2. Does the heart beat sensor work here?

Cheers,
York

ajs123, 2022-03-08:
Hi,

The original M3 computer measures the magnet position (resistance) in a different way which precludes having both connected at the same time. 

If you search online, you might be able to find an add-on box for the M3 that does just add bluetooth. The last time I checked (a couple of years ago), it was available only in Europe.

Hope that helps!

nealjane, 2022-03-08:
This guy sells them in Uk and ships worldwide - but then it only makes it into a standrd M3i (needing converter for apps like zwift) - 
https://lhproducts.company.site/

chanyork, 2022-03-09:
Hello,

Thank you for your email.

Yes, I just notice that it actually prevented me to connect both at the
same time.
Any adding of diode or resistor you think can help to achieve that?

Cheers,
York Chan

chanyork, 2022-03-09:
Hello,

I found out just disconnect the black color line to the original computer
than you can have both computer running.  and measuring the resistance.

Cheers,
York Chan
Mobile/Whatsapp  +852 98662380
***@***.*** ***@***.***>
http://www.petpalace.com.hk

ajs123, 2022-03-09:
That’s interesting. It confirms my suspicion that the Keiser computer is measuring the resistance intermittently. 

If you disconnect the black wire from the Keiser computer, you are using 3.3V from the feather as power to the sensor. This is likely to affect the calibration on the Keiser computer.

chanyork, 2022-03-09:
That is what I thought too.  But it did not.  Both Keiser and Kbikeble are
pretty in sync.  Hence it give me thinks that the Keiser is measuring %
changes in voltage drop.

By the way.
Can we add this in please!!!
1. ANT+ heart rate reading and send out via bluetooth as well
2. Screen to have Active Time Counter
3. Distance or Odometer I read that Keiser use 200 rap as 0.1 or 1 I can't
remember.

ajs123, 2022-03-11:
Keiser and KBikeBLE agreeing when connected this way might be a matter of good luck!

Regarding the other possible features...

1. ANT+  - It does appear that the nrf51 SoC supports ANT+ and BLE at the same time, so an interesting possibility.
2. Time counter - Should also be possible, as an option. Would need to make space on the display!
3. Distance/Odometer - It would be interesting if Keiser used something that simple. It's listed in the TODO as a future possibility. I was assuming that it would require a model, using power and cadence and maybe resistance...

nealjane, 2022-03-11:
See this related to Keiser speed and distance - 
https://github.com/ptx2/gymnasticon/pull/89

ajs123, 2022-03-11:
Thanks, @nealjane!  I’m a little surprised, but if Keiser and Peloton are meant to simulate bikes on flat ground maybe it makes sense to have a basic wind resistance model.

ajs123, 2022-03-11:
@nealjane the latest commit includes a speed calculation using these references.  I divided the coefficients by 0.6 to provide Km/h instead of mph. The calculation is only done if FTMS services is enabled since that's the only place it's used right now, and it's untested!

Thanks again!

closed 2022-03-16 by ajs123