ajs123/KBikeBLE #7: Issue to connect to IPAD/Iphone

chanyork, 2022-03-08:
I am having trouble to connect to zwift on ipad.  Can connect to the power via bluetooth but when click on cadence it found nothing.

ajs123, 2022-03-08:
Hi @chanyork,

A couple of questions...

1. Do you have both power and cadence showing on the display?
2. Does the power reading show up correctly in Zwift?

chanyork, 2022-03-09:
On the display yes.
Zwift shows connected to the power but always shows 0.
Zwift unable to connect to Cadence.

when I disable the CPS measurement it seems to be easier to connect.

//setupCPS();

Cheers,
York Chan

ajs123, 2022-03-09:
Hi,

Can you see what happens if you disable FTMS instead of disabling CPS? FTMS doesn’t provide a cadence; it provides a crank count and time and leaves it for the app to calculate cadence. And the speed (Km/hr) provided with FTMS is not correct. 

Let me know!

Alan

chanyork, 2022-03-09:
let me try that this few days.

chanyork, 2022-03-10:
Hello Alan,

Testing results as follow.
Device: Ipad Mini
Device: Iphone 13
Software: Zwift
FTMS enabled and CPS enabled - Cadence can not connect even bluetooth name not updated in Power but connected
FTMS enabled CPS disabled - OK
FTMS disabled and CPS enabled - Cadence can not connect even bluetooth name not updated in Power but connected
FTMS enabled and CPS enabled - Cadence can not connect even bluetooth name not updated in Power but connected

Software: Rouvy
FTMS enabled and CPS enabled - All Good.

I am not too sure what system Zwift is using but seems like it has issue with CPS.

ajs123, 2022-03-10:
To be sure I understand…

Everything works in Rouvy - both power and cadence coming through correctly?

When Zwift connects for power, is it still reading zero?

thanks!

chanyork, 2022-03-10:
Correct.

chanyork, 2022-03-10:
Yes, the below works.  I even removed the adddata part as well.

// setupFTMS();

at KBikeBLE.ino line 615,

try

//  Bluefruit.Advertising.addService(svc_ftms, svc_cps);   // Advertise the services
//  Bluefruit.Advertising.addData(0x16, FTMS_Adv_Data, 5); // Required data field for FTMS
  Bluefruit.Advertising.addService(svc_cps); // Advertise only the cps service

ajs123, 2022-03-11:
Just to be sure - Zwift works now?  If so, I'll disable FTMS at least temporarily.

chanyork, 2022-03-11:
Yeap Zwift and all other works.

ajs123, 2022-03-11:
Great!

I may have found the problem. The Fitness Machine Feature characteristic has Resistance level supported unset (because the app can't change the resistance like it could with a trainer). But in the Indoor Bike Data characteristic Resistance level present is set and the resistance is included among the data fields. I'd thought that it was OK to report the resistance even if the app can't change it.

As I re-read the spec, the resistance shouldn't be included in the Bike Data. I can give it a try, maybe this weekend.

ajs123, 2022-03-11:
I just committed changes to remove the resistance from the Bike Data characteristic. Not tested yet!

chanyork, 2022-03-11:
I just tried the new code.  It work for a short period and was back to the same situation.

ajs123, 2022-03-11:
Thanks. My advice in that case is to disable FTMS again. I'm not sure what
could be wrong but there's been so little testing with FTMS.

ajs123, 2022-03-11:
The latest commit provides options (options.h, around line 98) to separately enable CPS and FTMS services. On my test system (iPad), I have to disable FTMS to work with Zwift. 

Using CPS, Zwift sees both power and cadence and it calculates speed.

chanyork, 2022-03-12:
> 

yes I was pedaling occasionally to keep the Arduino on.  So at the moment I disable the FTMS.

ajs123, 2022-10-11:
So everything now works with Zwift?

chanyork, 2022-10-11:
Just tried the new code.  Does not seem to work on my side.  Still the same
results.

chanyork, 2022-10-11:
Yes with the removal on the line that was advices.

ajs123, 2022-10-11:
Thanks. That's good to know. At least for now, it seems that FTMS will need
to be disabled to work with Zwift.