ajs123/KBikeBLE #8: FTMS support

tamccauley, 2022-03-11:
This software is awesome.  I am quite new to programming.  In order to get this software to work with the RGT Cycling app, I disabled Advertising of the ftms service.  Apparently RGT software didn't like that?  Before I disabled it, my board would never connect.  After disabling, it connects every time.   If I decide to pay for Zwift, maybe I will have to re-enable?  Not sure.

Thanks

tamccauley, 2022-03-11:
I just made the changes you committed relative to removing resistance from FTMS Bike Data.  Now, RGT Cycling software connects to the your software, no problem.  No need to comment out the FTMS advertising, not that I really use it anyway.  Thanks for the fix.

ajs123, 2022-03-11:
Great!

ajs123, 2022-03-11:
In the latest commit, I added options to separately enable/disable FTMS and CPS. I had to leave FTMS disabled (comment out //#define PROVIDE_FTMS) to work with Zwift. Please let me know if RGT stops working for you.