ajs123/KBikeBLE #10: Pinout not consistent across M3 builds

tamccauley, 2022-03-18:
I got my Keiser M3 working with this code, modified slightly for my display and changes to the I/O.  It appears my bike's I/O is different from the pinout identified by KBikeBLE.  Below is what I am using for my bike and it is working with RGT Cycling app.  Thanks so much for this development.

My Keiser M3 Serial No: 080118  (which means built 2008 - Jan 18)
Keiser M3 interface through the RJ9 4p4c jack:

      Red     - CRANK_PIN - crank switch to ground (dry switch, not Hall effect)
               To a digital input. Transient protection is recommended.
      Yellow  - RESISTANCE_TOP - 10K magnetic brake position sense pot high side
               Use a digital output so it can be turned off to save power
      Green   - RESISTANCE_PIN - magnet brake position sense pot wiper
               To an analog input channel
      Black   - GROUND - magnetic brake position sense pot low / crank switch low side

ajs123, 2022-03-18:
Thanks for noting this! The original test bike is from 2012. 

Did you tap into the Keiser cable, or are you using a standard RJ9 handset cable? If a Keiser cable, are you able to see the colors from left to right in the RJ9 plug?

For now, I updated the HOWTO with a recommendation to check the colors as described at the end of the document.

In any event, it's so nice to see this working for other people!

tamccauley, 2022-03-18:
I ordered a 4p4c RJ9 connector and cable from digikey.  It has the exact same color coding as the cable on the bike.  To use the system, I disconnect the original cable from the jack by the flywheel and plug my cable in there.  Black Red Green Yellow, left to right looking at the connector with the tab facing me is the orientation.

ajs123, 2022-03-18:
Thanks.  

[This picture of the pickup assembly](https://www.sportsmith.com/speed-pickup-assembly-with-circuit-board-keiser/product/46112?serial=1187) shows Black-Red-Green-Yellow as well, and following the traces on the photo of the back indicates Pot_high - Pot_wiper - Crank_sw - Common. 

According to the parts listings at Sportsmith, the parts are different but unfortunately the photos are all of the same part.

I can further update the HOWTO.  Thanks again!