ajs123/KBikeBLE #11: Photos of hardware connection

SamuelMatson, 2022-04-16:
Thanks for this awesome project. It would be awesome if you could include or link to photos and instructions for the hardware connection on the M3. I don't have an M3 yet so I want to know what's involved with this before I make the purchase.

ajs123, 2022-04-16:
For now, there are some photos in build.md that might help. In that example, I used the cable that’s in the M3 including the connector that plugs into the M3 computer. 

Please be aware that it seems that Keiser may have changed the pin assignments on the RJ9 connector on the bike. See https://github.com/ajs123/KBikeBLE/issues/10. If you go to sportsmith.com or anyplace that sells parts, and choose the sensor board, you’ll see the serial number at which they may have changed.   It’s not hard, though, to tell which pin is which as long as you have an ohm meter.

SamuelMatson, 2022-05-20:
Hi,

This is a late reply but just wanted to say thank you for your help.

Cheers,
Samuel

closed 2023-01-24 by ajs123