ajs123/KBikeBLE #15: Featherwing OLED boot loop

antknight, 2023-01-16:
Hi,

Thanks for this awesome project! 

I went ahead and purchased the Adafruit feather and featherwing OLED 128X64 as mentioned in [HOWTO.md](https://github.com/ajs123/KBikeBLE/blob/main/HOWTO.md). There is very little information available regarding using the 128X64 SH1107 OLED with U8G2 and I am struggling.

I have altered the U8G2 constructor from:
`U8G2_SH1106_128X64_NONAME_F_HW_I2C display(U8G2_R1, /* reset=*/U8X8_PIN_NONE);`
to:
`U8G2_SH1107_64X128_F_HW_I2C display(U8G2_R0, U8X8_PIN_NONE);`

Without the display plugged in, it works as expected BLE etc. When I plug the display in, as soon as the display shows "KBikeBLE" followed by the version, the feather reboots? in a loop: [GIF](https://gfycat.com/orderlyhonoreddragon)

I am not really sure where to start debugging. I tested the display using the code on [this page](https://learn.adafruit.com/adafruit-128x64-oled-featherwing/arduino-code) and it works fine so no issues with the hardware.

Any help much appreciated.

ajs123, 2023-01-16:
Hi @antknight ,

Display anomalies can come from using the wrong constructor. But the constructor  you're using looks right to me. In another project, with that same featherwing display mounted in "landscape" rotation, the correct constructor is 

`U8G2_SH1107_64X128_F_HW_I2C display(U8G2_R1, U8X8_PIN_NONE);`

so the _R0 ought to be right for you. And the wrong _Rx should just yield incorrect rotation of the display. 

Behaving differently on battery suggests a hardware problem. Some thoughts:

1. Just in case it's something with the featherwing, can you try one or more of the U8G2 library examples? Hello_world in the Full Buffer group, then the Graphics Test, may be helpful.
2. More likely... after displaying "KBikeBLE" and starting bluetooth services, the startup code sets the crank switch pin to be an input then applies +3.3V to the resistance sense potentiometer (RESISTANCE_TOP).  Is it possible that that pin is shorted to ground (or connected to ground via something substantially less resistance than the bike's 10K ohm pot? 

Let me know what you discover!

antknight, 2023-01-17:
I have done a bit more testing by separating the Feather and Featherwing OLED and placing them on breadboard with only VCC, GND, SDA and SCL connected (I don't have the bike sensors connected at this stage)

- The display works perfectly with the Adafruit example code.
- The U8G2 Hello_world Full buffer example doesn't work. Hello World flashes briefly on the screen then nothing else happens and it doesn't appear to be in a boot loop.

So it seems like this is a compatibility issue between the Featherwing OLED and U8G2.

I should probably raise an issue over on u8g2. I have ordered a separate 1.3" OLED as used in your example.

antknight, 2023-01-18:
In that case, I will order another Feather + Feather OLED in case there is some oddity with the bits I have. I will let you know how it goes.

ajs123, 2023-01-18:
Just in case of some recent issue with U8G2, you could first try reverting to version 2.32.15 or earlier of that library. I've found that library to be very reliable and professionally maintained but there's always a chance...

Other possible things to check before buying stuff...

- Solder joints at 3.3V and ground. The Adafruit and U8G2 libraries might set up the display chip a bit differently, so conceivably they could have different tolerances for power issues.
- Re the "boot loop" - on the Featherwing while disconnected from the Feather, check that there is no conductivity (or at least very high resistance) between RESISTANCE_TOP (see bike_interface.h) and 3.3V or ground. This pin somehow getting bridged to ground would definitely cause the boot loop.

antknight, 2023-01-19:
OK so, new displays arrived.

- The 1.3" OLED works perfectly
- The new feather + new featherwing OLED works perfectly
- The new feather + original featherwing OLED same old problem

I had a probe around with a DVM trying to find differences between the two sets of components with no success.

At this stage as I now have a working solution I think we can put it down to a hardware issue and close this ticket. Thanks for your help with this and thanks for creating such a useful project!

closed 2023-01-19 by antknight