ajs123/KBikeBLE #16: Stuck on log screen when FTMS enabled

antknight, 2023-01-29:
I would like to try BitGym which I believe requires FTMS. When I enable FTMS by uncommenting #define PROVIDE_FTMS in options.h, it freezes on the log screen during startup. 
[image]

ajs123, 2023-01-29:
Hmmm... First question: Does bluetooth start (does the blue LED begin
flashing?)

antknight, 2023-01-29:
No, no connection via bluetooth possible.

ajs123, 2023-01-29:
I'm not in a good position to try to replicate on this end. Would you be
able to try release v1.1 instead of main?

antknight, 2023-01-29:
I can confirm v1.1 works and is detected by BitGym!

ajs123, 2023-01-30:
Thanks. I added a note in the README about this being an issue in the main branch and added the fix to the TODO.

closed 2024-01-07 by ajs123