ajs123/KBikeBLE #20: Transient suppression on Crank Pin

tamccauley, 2023-12-16:
I have failed a couple Crank Pin inputs on my processors. I do not have any suppression, despite your recommendation.  I looked at the signal with a scope and it looks very clean, no voltage spikes.  Did you have problems with this input before you added surge suppression?  My project has run for hours before actually failing the input previously.  Today, it failed after only 5 minutes so trying to find a solution that I can be confident with.  Wondering if your suggestion for suppression was based on experience on this project or experience with reed switches in general or just good practice.

Thanks again for the awesome project.  I actually have this running on a Seeed XIAO NRF52840 as it is much smaller and cheaper than the Adafruit Feather you used.

ajs123, 2023-12-16:
I never tested without any protection for the input, as I assumed it would be a problem with a reed switch at the end of a longish cable. The small series resistor (which relies upon there being some capacitance at the input) was the only thing I tried. 

There will always be a transient when the switch opens, the energy being dependent upon Vdd, the pull-up on the input, and the inductance of the cable. It might be hard to see depending upon the speed of your scope. 

It’s cool that it works with another, economical, processor board. What Arduino libraries are you using? There are some calls to the RTOS and I don’t know how specific they are to the Adafruit implementation.

tamccauley, 2023-12-16:
I pretty much used your code. I had to make the below mods.  Thanks for the feedback.

Added “static const uint8_t VBAT  = PIN_VBAT ;” to Variant.h file so I could use this in bike_interface.h.
In Bike_interface.h, instead of #define BATTERY_PIN A6, changed to #define BATTERY_PIN VBAT

closed 2024-01-07 by ajs123