ajs123/KBikeBLE #23: Disable change shifting/resistance by moving the resistance lever to the top

serbasan, 2024-10-23:
Hello. 

I want only gears 1-24 to be displayed. How can I disable the change to the 1-100% mode? 

It happens during training that I’ve moved the lever all the way up, and the display changes. It’s a bit annoying because I’m very used to the 1-24 gears. 

I would appreciate knowing which part of the code I should modify. 

Thank you very much.

ajs123, 2024-10-23:
Hi,

In KBikeBLE.ino, you can comment out lines 811 and 812 as follows (actual line numbers will be different if you're using another release):

//    gear_display = !gear_display;
//    rg_label = rg_labels[(int)!gear_display];

serbasan, 2024-10-24:
Thank you very much, I have changed it now. I am really enjoying your project with SmartSpin2k. They work together wonderfully with my training and racing. Are you still using your Keiser M3? I hope so. Best regards.

closed 2024-10-24 by serbasan

ajs123, 2024-10-24:
I am. I’m a percentage person. My wife is a gear person. Thus the feature.
This should really be a configuration option!