ajs123/KBikeBLE #24: What can this connect to?

travisvanneste, 2024-12-12:
This will be my first arduino project, so forgive the newbie question..

I'm looking to connect this to Wahoo fitness app (or anything else really) - just wondering if this is going to send both the power data and cadence out via bluetooth...  The end goal of course is to work with something like zwift, but at first just to be able to log data.

Wahoo appears to look for bluetooth sensors, just not sure if it will recognize this as a sensor?

ajs123, 2024-12-12:
KBikeBLE appears to apps as a standard bluetooth power meter, providing power and cadence. Much of my personal use has been with Wahoo X and it worked just fine.

travisvanneste, 2024-12-12:
That’s great!  Looking forward to bringing Bluetooth connectivity to two
old M3 bikes that I own.

You did great work putting this together!

 I may 3d print an enclosure to make the new computer attach to the bike….
Or I may try and use the old display as it looks like someone else who
commented on this did…. Really all I’m interested in is the Bluetooth
functionality.

Thanks again for your great work!

Travis W. Van Neste
Van Neste Surveying
3845 40th Ave South
Minneapolis, MN 55406

952-686-3055

travisvanneste, 2024-12-30:
So I got this working with Wahoo - reading power and cadence - any way for
it to get 'speed' to wahoo?  I'm sure the Keiser computer gets its speed
from cadence / power?