ajs123/KBikeBLE #25: Shouldn't his work without the display?

travisvanneste, 2025-01-22:
I build two of these - 1 without the display because I was really only interested in using the bluetooth functionality.  I build a small enclosure with a short phone cord and rj9 connector on one end and a phone jack on the other end so that the original computer still works (connected all wires internally except the black, per recommendation of another user).  Seems to work pretty good.

The one without the display does not work though  - it connects to bluetooth but shows zero watts and zero rpms.  Also the keiser computer will not work when hooked up to it.  I cannot find anything different except not having the display wired.

Any suggestions?

ajs123, 2025-01-22:
I'm not certain what it will do without a display. The display is initialized before bluetooth, so your experience suggests that at least none of the functions called in display_setup() hang in the absence of a display.

One thing you could try, short of temporarily adding a display: Start but don't connect to bluetooth, and wait for the blue LED to go dark. Then turn the crank and see if it wakes up. That would demonstrate that crank sensing is happening.

robbybara, 2025-08-01:
I was thinking about this as well. I would like to keep the current screen and just intercept the data and provide a pass-through so that it just uses the screen that it already has. I was thinking this must be possible.

ajs123, 2025-08-02:
Here’s the challenge: