ajs123/KBikeBLE #26: Question about simultainious display use

magpern, 2025-04-20:
I'm doing a similar project for older BodyBike spinn bikes and I got the question if I can support Keiser M3.
My question is, would it be possible to get the data from the Keiser M3 by "man in the middle" approach?
Goal is to still have the old display on the bike, but just get the data from the bike and transform it into BLE (and/or ANT+)

ajs123, 2025-04-20:
It’s possible but would require some study of how Keiser measures the resistance magnet position and design of corresponding electronics or software approach. Keiser doesn’t maintain a fixed voltage at the potentiometer. Most likely (to save battery), they apply the voltage briefly each time the computer takes a measurement. So it may involve, for example, a peak hold on the wiper voltage.

magpern, 2025-04-20:
Thanks for replying. I saw there is(was) some device you could buy as a "man in the middle" thing that converted the bike to bluetooth, but quite expensive.
In my head a Y-connector from sensor, one leg of the Y feeding the display, then the other leg of the Y to an nRF52840 (and some resistors in between according to chatGPT). Don't know if it could be that simple, then one could monitor the peak of the wiper voltage. The reed sensor should be fine.
I'll guess I'll just have to try.

closed 2025-04-20 by magpern