ajs123/KBikeBLE #28: Compatibility List / Check

Sc0ttmeup, 2025-07-11:
Is it possible to get a known compatibility list along with any caveats for M3 models, or a way to check if models are compatible with KBikeBLE?

I'm pretty sure the manufacture date on mine is in November 2021, any idea if KBikeBLE will work?

- Model: 005502BBC
- Serial No: 211116-xxxxx

nealjane, 2025-07-12:
M3bike from late ~2015 onwards just needs a computer replacement (m3i computer/or M-connect) to be upgraded to full m3i with bluetooth. (Serial states manufacture date eg YYMMDD)

ajs123, 2025-07-12:
KBikeBLE was intended for the M3. As far as I can tell (I looked at replacement parts [here](https://sportsmith.com/parts-by-brand/keiser/bikes/m3i/5501xxx/?_bc_fsnf=1&Item+Type=Speed+Sensors)), the crank speed and resistance pickup board on the M3i is the same, so it ought to work. That's consistent with @nealjane's reply noting that the M3i computer will work with a late model M3.

Sc0ttmeup, 2025-07-18:
I confirmed that the M3i M Series / with M Connect Computer is compatible with the part listed.

@nealjane if I understand you correctly, a M3i manufactured in November 2021 will be compatible with KBikeBLE?

[sportsmith keiser m3 Speed Pickup with Circuit Board, Keiser, 550817 _ 550817, 550904, 550869.pdf](https://github.com/user-attachments/files/21316625/sportsmith.keiser.m3.Speed.Pickup.with.Circuit.Board.Keiser.550817._.550817.550904.550869.pdf)

[image]